VladimirP1/esp-gyrologger #2: LANC recording

VR-VIDEO, 2022-07-20:
It will be awesome if  pressing  button on logger  its also can send LANC signal to camera  start recording.

VladimirP1, 2022-07-20:
Should be doable. LANC also has 5-8v power supply in the connector, this would be useful for powering the logger.

mmstac, 2022-09-12:
Here are some details on the LANC protocol, I believe it is 5.8 volts?
Many years back I used an IR LED to remote trigger cameras, I had the working protocol for Sony NEX, Canon, Nikon and some other cameras. It was written for Arduino, but I assume only timing is important. If you are interested to integrate that, I can dig up the file. BTW thanks for all your work on ESPLog, it works wonderfully!!

http://www.boehmel.de/lanc.htm

VladimirP1, 2022-09-12:
Are there any actual cameras which are worth using with GyroFlow and support LANC?
I could only think about z cam e2...

Anyway, if this is really useful, I can try adding it, but only of I find someone with an actual LANC camera, who is willing to test. And this will also require some additional circuit (like 1 transistor or something).

cabinlab, 2022-11-20:
> Are there any actual cameras which are worth using with GyroFlow and support LANC? I could only think about z cam e2...
>
> Anyway, if this is really useful, I can try adding it, but only of I find [...]

I have a zcam e2, and LANC would help because I already have to use the I/O port for camera control from the gimbal. I have extra esp32 modules, but It will take a couple weeks to get an IMU to my location.

A more perfect solution for me would be to trigger the esplog from the camera, or from the gimbal (DJI RS2 has documented protocol), but that's a new issue.

VladimirP1, 2022-11-20:
Triggering EspLog from zcam could be possible over wifi.

Check what your browser shows when you navigate to http://<IP address of z-cam>/ctrl/mode?action=query

VladimirP1, 2022-11-20:
I'd like to know if it works without an active session

VladimirP1, 2022-11-20:
There's also a mention of some kind of websocket notification interface, but this https://github.com/imaginevision/Z-Camera-Doc/blob/master/E2/protocol/http.md#Asyc-notification . If you can save the web page from the camera they are talking about, this could be useful

VladimirP1, 2023-06-14:
Not implemented at the moment. But receiving seems to be easier than sending LANC commands - just using UART at 9600. If you find the an exact byte sequence produced when the cam is triggered, I can add support for this.

VladimirP1, 2023-06-14:
Probably something like FF 03

VladimirP1, 2023-06-14:
The problem with the display is that I have used u8g2 for drawing, which supports only monochrome displays.
Adding support for color lcd's is not currently planned as I do not have time for that and even for more useful things like changing the binary logging format to a different one understandable by gyroflow without conversion step.
I will probably look into triggering logging over lanc, seems not that hard to implement

VladimirP1, 2023-06-20:
I am on a trip right now, I will be back on Wednesday.
It won't work with standard software because lanc is constantly polling your trigger source for commands.

VladimirP1, 2023-06-23:
> > I am on a trip right now, I will be back on Wednesday.
> > It won't work with standard software because lanc is constantly polling your trigger source for commands.
> 
> Great grinning👍🏻

 Try the latest build from actions. You will need to set cam control type to 6 and connect LANC from the y splitter to cam control pin.

VladimirP1, 2023-06-24:
> > > > I am on a trip right now, I will be back on Wednesday.
> > > > It won't work with standard software because lanc is constantly polling your trigger source for commands.
> > > 
> > > 
> > > Great grinning👍🏻
> > 
> > 
> > Try the latest build from actions. You will need to set cam control type to 6 and connect LANC from the y splitter to cam control pin.
> 
> Thanks. Tested it but it doesn't works. Camera records with the LANC signal but connecting it to one of the pins and setting it up doesn't triggers the logger

Are you connecting the camera, button and the logger through a Y splitter? Or are you just connecting the logger to the button?

VladimirP1, 2023-06-25:
You are doing everything correctly I think except there is no point in connecting the remote with a single cable - in LANC the camera initiates all communication so if there is no camera, there will be no communication.

See if this works [esp-gyrologger.zip](https://github.com/VladimirP1/esp-gyrologger/files/11859646/esp-gyrologger.zip).
I changed it to react to any LANC command, because I do not know what command does this particular remote send.

 If it does, note the commands that are actually being sent in the output from the usb virtual com port which appears if you connect the atoms3 via usb to your pc.

handcraftedcc, 2025-02-26:
Could this be integrated into the logger?
https://github.com/gyroflow/flowshutter/blob/master/tools/LANC.ino