VladimirP1/esp-gyrologger #3: control rgb led?

gretel, 2022-08-02:
the [m5stamp](https://shop.m5stack.com/products/m5stamp-c3u-mate-with-pin-headers) boards have a `SK6812` led - the form factor is pretty great for the logging application. am not good with `c++`.. found a library at least: https://github.com/UncleRus/esp-idf-lib/tree/4ca15c4a1495a65def31ed83ef905fba67dde18f/components/led_strip 😄

VladimirP1, 2022-08-02:
Ordered some ws2812b, will see if this works

VladimirP1, 2022-08-03:
Here's the official esp-idf example code for ws2912 https://github.com/espressif/esp-idf/tree/master/examples/peripherals/rmt/led_strip

gretel, 2022-08-03:
@B-nutze-RR i'm trying to have a segment of https://de.aliexpress.com/item/4000154483882.html in the meantime - it's 5V and bright.

VladimirP1, 2022-08-04:
[image]

VladimirP1, 2022-08-04:
code is in master

closed 2022-08-04 by VladimirP1

gretel, 2022-08-04:
> code is in master

awesome!

B-nutze-RR, 2022-08-04:
Very Nice 💪

gretel, 2022-08-04:
looks and works great using a [`m5stamp c3u`](https://docs.m5stack.com/en/core/stamp_c3u), pins are as shown:

[image]