VladimirP1/esp-gyrologger #4: orientation issue?

gretel, 2022-08-09:
not sure if i get the code right, but is the right orientation written to the log file?

this is the header of the logfile, right?

https://github.com/VladimirP1/esp-gyrologger/blob/5f984de0906d5947b01854407623eb07cacc70f9/main/wifi/http.cpp#L87

here it gets initialized to `xyz` for the webinterface?

https://github.com/VladimirP1/esp-gyrologger/blob/53596aa058caa5491977f8bf8b3c6ecb3fec6268/main/storage/settings.hpp#L57

what's happening here?

https://github.com/VladimirP1/esp-gyrologger/blob/4415ae175a72ad01c2af83479042a277aed7e55f/main/wifi/http_strings.hpp#L984

VladimirP1, 2022-08-09:
`respond_with_file` is not actually used anymore, it is the code which does log decompression on the logger.

The log decompression code has been moved to the client-side (this long base64 encoded string in `http_strings.hpp` is the WASM code for decompression)

VladimirP1, 2022-08-09:
Here
https://github.com/VladimirP1/esp-gyrologger/blob/4415ae175a72ad01c2af83479042a277aed7e55f/main/wifi/http_strings.hpp#L984
the IMU orientation which has been set in settings gets inserted into the client-side JS code

VladimirP1, 2022-08-09:
`xyz` is just the default value

VladimirP1, 2022-08-09:
As the orientation is inserted directly in JS code, be sure to reload the page with the controls after changing the orientation in settings

closed 2022-08-11 by VladimirP1

gretel, 2022-08-16:
@VladimirP1 thanks for all the info!

still, having issues getting the orientation right in gyroflow. i can't mount the `imu` in flush with the diagram printed on the pcb, it's actually tape tucked to the side of the camera.

i've tried changing the orientation in gyroflow to what i assume is correct but i don't get it to sync any.

VladimirP1, 2022-08-16:
Have you tried 'guess IMU orientation' in the timeline context menu?

Anyway, you can upload your files somewhere and I will have a look.

You can DM me on discord (you can find me as `@vpinch4` in GyroFlow discord) or telegram `@vpinch`