VladimirP1/esp-gyrologger #6: Maybe unstable

user1321, 2022-09-06:
Good day!
First of all thx for great project.

Hardware:
MPU-6050 +  ESP-WROOM-32D 16МB + led+ resistor + button
Take attention board: 16MB

Firmware:
Ready to use firmware from lates build for ESP32.

Problem description:
Board was working fine for 3 days. After 20-50 use times it start to reboot itself after hardware switch was pressed or in case of click  at red circle inside WEB GUI.
UART Logs will be added.

What I checked:
1. Firstly I was thinking I cookup\burned IMU or\and ESP. 
I soldered to different one`s GPIO IMU and checked it inside WEB calibration page. IMU successfully had transferred data to calibration circule.
2. Tried to reflash firmware again. Got same error + restart (please, check logs).
3. Compile firmware from sources using "idf 4.4". - no luck. same result - reboot.
4. Workaround was to erase flash using:
idf.py -p com3 erase_flash

Only after that after firmware was uploaded again board started to operate as expected. (no reboots + new log file was created)

Thoughts:
1. Maybe it is board incompatible. 4MB vs 16MB. If it is - all fine.
But if that:
2. When I was trying to compile source code using "idf 5.0" it stoped with critical warning (? was thinking only errors are critical) about that line of code can write data to different place in memmory (something like overflow). If I understand correctly.

 But, unfortenatly I did`t print screen. Will install IDF 5.0 and will compile again.

Two logs: (1 - reboot\crash, 2 - normal):

1:
I (27) boot: compile time 12:04:18
I (27) boot: chip revision: 1
I (29) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (36) boot.esp32: SPI Speed      : 40MHz
I (41) boot.esp32: SPI Mode       : DIO
I (45) boot.esp32: SPI Flash Size : 4MB
I (50) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (74) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (81) boot:  2 factory          factory app      00 00 00010000 00100000
I (89) boot:  3 storage          Unknown data     01 81 00110000 002f0000
I (96) boot: End of partition table
I (100) boot_comm: chip revision: 1, min. application chip revision: 0
I (108) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=2e1d0h (188880) map
I (185) esp_image: segment 1: paddr=0003e1f8 vaddr=3ffb0000 size=01e20h (  7712) load
I (188) esp_image: segment 2: paddr=00040020 vaddr=400d0020 size=7e50ch (517388) map
I (377) esp_image: segment 3: paddr=000be534 vaddr=3ffb1e20 size=06c88h ( 27784) load
I (389) esp_image: segment 4: paddr=000c51c4 vaddr=40080000 size=16f44h ( 94020) load
I (427) esp_image: segment 5: paddr=000dc110 vaddr=50000000 size=00010h (    16) load
I (440) boot: Loaded app from partition at offset 0x10000
I (440) boot: Disabling RNG early entropy source...
I (452) cpu_start: Pro cpu up.
I (452) cpu_start: Starting app cpu, entry point is 0x400815b4
I (0) cpu_start: App cpu up.
I (468) cpu_start: Pro cpu start user code
I (468) cpu_start: cpu freq: 160000000
I (468) cpu_start: Application information:
I (473) cpu_start: Project name:     esp-gyrologger
I (478) cpu_start: App version:      1
I (483) cpu_start: Compile time:     Aug 30 2022 12:04:11
I (489) cpu_start: ELF file SHA256:  cb947bc249ad64e3...
I (495) cpu_start: ESP-IDF:          v4.4.1
I (500) heap_init: Initializing. RAM available for dynamic allocation:
I (507) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (513) heap_init: At 3FFBE1C8 len 00021E38 (135 KiB): DRAM
I (519) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (526) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (532) heap_init: At 40096F44 len 000090BC (36 KiB): IRAM
I (539) spi_flash: detected chip: generic
I (543) spi_flash: flash io: dio
W (547) spi_flash: Detected size(16384k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (561) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (758) wifi:wifi driver task: 3ffc8368, prio:23, stack:6656, core=1
I (758) system_api: Base MAC address is not set
I (758) system_api: read default base MAC address from EFUSE
I (758) wifi:wifi firmware version: 63017e0
I (768) wifi:wifi certification version: v7.0
I (768) wifi:config NVS flash: disabled
I (768) wifi:config nano formating: disabled
I (778) wifi:Init data frame dynamic rx buffer num: 32
I (778) wifi:Init management frame dynamic rx buffer num: 32
I (788) wifi:Init management short buffer num: 32
I (788) wifi:Init dynamic tx buffer num: 32
I (798) wifi:Init static rx buffer size: 1600
I (798) wifi:Init static rx buffer num: 10
I (798) wifi:Init dynamic rx buffer num: 32
I (808) wifi_init: rx ba win: 6
I (808) wifi_init: tcpip mbox: 32
I (818) wifi_init: udp mbox: 6
I (818) wifi_init: tcp mbox: 6
I (818) wifi_init: tcp tx win: 5744
I (828) wifi_init: tcp rx win: 5744
I (828) wifi_init: tcp mss: 1440
I (838) wifi_init: WiFi IRAM OP enabled
I (838) wifi_init: WiFi RX IRAM OP enabled
I (848) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (948) wifi:mode : softAP (40:91:51:b1:65:c9)
I (948) wifi:Total power save buffer number: 16
I (958) wifi:Init max length of beacon: 752/752
I (958) wifi:Init max length of beacon: 752/752
I (1698) wifi:Total power save buffer number: 16
I (1698) wifi: wifi_init_apsta finished.
I (1708) storage_fat: Mounting FAT filesystem
I (1718) gyro_ring: Loaded accel offsets 0 0 0
I (1718) gyro_ring: Gyro calibration loaded
I (1728) gyro_ring: inv_interval 0.000556
I (1728) gyro-prober: Probing 68
I (1778) gyro-prober: Probing 68
I (1828) gyro-prober: Probing 68
I (1878) gyro-prober: Probing 69
I (1928) gyro-prober: Probing 69
I (1978) gyro-prober: Probing 69
I (2028) gyro-prober: Probing 6A
I (2078) gyro-prober: Probing 6A
I (2128) gyro-prober: Probing 6A
I (2178) gyro-prober: Probing 6B
I (2228) gyro-prober: Probing 6B
I (2278) gyro-prober: Probing 6B
I (2328) gyro-prober: Probing 68
I (2328) gyro_mpu: IMU reset
I (2328) http-server: Starting server on port: '80'
I (2348) http-server: Registering URI handlers
I (2428) gyro_mpu: IMU wake up
I (2428) gyro_mpu: IMU change clock
I (68218) wifi:flush txq
I (68218) wifi:stop sw txq
I (68218) wifi:lmac stop hw txq

abort() was called at PC 0x401391a6 on core 1

Backtrace:0x40081e96:0x3ffd5dd00x4008b7a5:0x3ffd5df0 0x40091c16:0x3ffd5e10 0x401391a6:0x3ffd5e80 0x401389bf:0x3ffd5ea0 0x4013867f:0x3ffd5ec0 0x400d7f9e:0x3ffd5ee0 0x400d96d7:0x3ffd5f10

ELF file SHA256: cb947bc249ad64e3

Rebooting...
ets Jun  8 2016 00:22:57
____________________________
2: (normal) :
......
I (1735) gyro_mpu: IMU wake up
I (1735) gyro_mpu: IMU change clock
I (4465) wifi:new:<11,0>, old:<11,2>, ap:<11,2>, sta:<255,255>, prof:11
I (4465) wifi:station: 14:16:9e:1e:f1:3f join, AID=1, bgn, 20
I (4625) wifi: station 14:16:9e:1e:f1:3f join, AID=1
I (4845) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
W (5475) wifi:<ba-add>idx:2 (ifx:1, 14:16:9e:1e:f1:3f), tid:0, ssn:38, winSize:64
I (13135) http-server: Start logging!
I (13445) logger: Using filename: /spiflash/LAA00001.bin
I (13445) logger: Opening file /spiflash/LAA00001.bin
I (13545) logger: 591 bytes, capacity = 17.64 h, max_error =  0.0560
I (13545) logger: 24 bytes of accelerometer data
I (13915) logger: 152 bytes, capacity = 2.03 h, max_error =  0.0111
.....
P.S. About  ELF file SHA256: cb947bc249ad64e3... I don`t remember if that log from original firmware or from compiled one, but error was same. Only address is different.

VladimirP1, 2022-09-06:
Will try to reproduce that build and decode the backtrace...

user1321, 2022-09-06:
About IDF 5.0 - ok, got it. I will not use it.
For next couple of days I will do some test of your current pre-build firmware.
In case if I will receive same error, I will let you know.
Thx.

user1321, 2022-09-06:
Or it is better to compile and flash from sources to be able to do use "https://github.com/me-no-dev/EspExceptionDecoder" or something like that? In any case will see if that will heppens again.

VladimirP1, 2022-09-06:
Okay, I've rebuilt the fw with using the same IDF docker image. 

```0x401389bf
/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_throw.cc:90
0x4013867f
/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/src/c++11/functexcept.cc:74 (discriminator 2)
0x400d7f9e
/opt/esp/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/ext/string_conversions.h:83
0x400d96d7
/opt/esp/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/bits/basic_string.h:6411
```

I've looked at `string_conversions.h` code, it looks that it is related to std::stoi somehow...
So I guess it is this line https://github.com/VladimirP1/esp-gyrologger/blob/master/main/logger/logger.cpp#L39

VladimirP1, 2022-09-06:
Probably the filename of one of the logs got corrupted after a power cut which caused some random characters in it. So stoi failed

VladimirP1, 2022-09-06:
This needs to be fixed - probably by ignoring these strange filenames here and deleting them automatically in the code.

By the way, you can use all 16MB of your flash space by changing the `storage` partition size in `partitions.csv`

user1321, 2022-09-06:
Sure, thx for advice about 16MB. that was the goal. I just was thinking to start with original firmware without modifying anything to see how it have to work)
Good catch about  "int idx = std::stoi(filename.substr(3, 5));"

VladimirP1, 2022-09-07:
I think I have fixed this

closed 2022-09-07 by VladimirP1