VladimirP1/esp-gyrologger #10: hang on powerup

mmstac, 2022-09-19:
Hi, one of my 3 Esploggers is hanging on powerup. The LED will come on, and they stays solid. There is no response/wifi.
I did not experience this with my other ones. If I replug the power then usually it boots correctly on the second power up.
Did you see this behavior before, is there a pin I should try swapping, or do I need to pull up the EN pin?

The hardware is the mini 01M module powered by BMI160 board with the following config.
1k resistor is added on the camera pin and a 160 resistor with red LED on LED pin with following setup.

[image]

VladimirP1, 2022-09-19:
The LED lights up, I think that means that the firmware has started up, but I can't understand what happens then...
Hard to imagine how can a hang happen, because esp-idf sets up multiple watch dogs which are pretty hard to disable.

VladimirP1, 2022-09-19:
If you can consistently reproduce it, try loading this firmware and connecting a USB-uart convertor to gpio21 at 115200 baud. (yes, an external one, because that way you can open the COM port before powering up the logger).
[esp-gyrologger.zip](https://github.com/VladimirP1/esp-gyrologger/files/9599962/esp-gyrologger.zip)

You can update OTA without soldering USB:
You need to extract the zip and upload the bin file on settings page. When the upload succeeds, wait for ~30 seconds for the update to finish and reconnect power.

Warning: settings will be reset, you can leave the tab with settings open and click on the apply buttons after reconnecting to wifi.

VladimirP1, 2022-09-19:
GPIO21 is labeled as TX on this board

mmstac, 2022-09-19:
Thanks, ok this logger is "embedded" inside a whoop and otherwise doing its job nicely. I have ordered some spare modules, when they arrive I will swap out and test the debug firmware. My other thought is maybe the regulator on this particular BMI160 is not providing sufficient current? The bare esplog draws about 90ma and without wifi 35ma, but maybe more at bootup.
The LED pin should be low before the firmware starts?

VladimirP1, 2022-09-19:
As I understand, gpio8 should be in high-impedance state by default, if it if high, then it means that the firmware was loaded and had a chance to execute some code (until the point where gpio8 is initialized).

Yes, it draws more at bootup and the issue might be due to the regulator not supplying enough power. There's an RF calibration phase during which there is a short (i guess <2ms) pulse of ~500ma power consumption. But I think this would also look like a boot loop rather than a hang.

closed 2022-10-08 by VladimirP1