VladimirP1/esp-gyrologger #11: Improvements

eirasys, 2022-10-05:
Hi, good work!
I'm testing your code with Lolin mini ESP32 C3 and BMI160.
How can we change the gain of the accelerometer and also the gyro gain?
I'm trying to stabilize video from DJI FPV that has no gyro data available.
So I'm acquiring it directly from the BMI160 glued to the camera.
Add some navigation links to the initial html page, to go to /settings and /calibration
Can't understand the green button! What it does?
Also add IP address to the log so that we can easily find the device.
Thank you,
João Antunes

VladimirP1, 2022-10-05:
Why would you want to change the gain?
If you want to correct for gyro gain error, how would you know the correct gain?

The green button does gyro calibration (you need to keep the board still when pressing it).

Cannot understand your suggestion about the IP address. IP address of what?

eirasys, 2022-10-05:
Made a small test rotating, toll left, roll right, pitch up, pitch down.
The result:
[image]

Low range and bad resolution in accelerometer. Is this what it's supposed to read?
Can't make any stabilization because the amplitude of the signal is too small and the result is noise added to the camera motion.

Not that important with the IP address of the ESP32, you could output the info to the console when it connects the WiFi network.

[image]

VladimirP1, 2022-10-06:
Resolution is enough and the scale is correct. You are probably using GyroFlow incorrectly. You need to find your IMU orientation and add some sync points.

1. Calibrate gyro (needed only once, saved to flash)
2. Record a test video.
3. Find the rough offset manually (for example, you can pick your drone from the ground, this event is easy to see in both video ang gyro data)
4. Right-click the timeline at a point where there is some significant motion on all axes, click "Guess IMU orientation". Or figure out the orientation manually. You can now save it in esplog settings so you do not have to enter it manually every time.
5. Add some sync points

https://youtu.be/5l1OEMuwN60

Did you do all these things?

If it still does not work, I can have a look at your files.

Okay, understood the point about Wi-Fi, that you are using it in station mode.

eirasys, 2022-10-07:
Thank you! You were right. I followed your video and I was making several mistakes.
Initially I synced the video hitting the camera and tested also with the start of the motors.
Tried to understand the IMU orientation, the BMI160 has two sets of axes drawn in the board, can't understand why. Tested [yzX]! To difficult things the board is glued upside down the camera. Then I followed the chip orientation as described in its manual. And to match the orientation axis of Gyroflow [yZX], also no success. The best solution is to let it "Guess IMU orientation", testing it in several places until one of the combinations succeed, the [Zyx].
Only now I could "Auto sync here" to fine tune the time offset.
I have big offset delays of more than 1s because I only start the logger after start recording the video.
To facilitate setting the "Rough gyro offset" there could be a timestamp or semaphore in the logger screen so that we could show the phone screen to the camera in the beginning and end of the movie. But then we would have a delay of wifi. Better alternative would be to use a led connected to the logger that we can put in front of the camera. It will light for a start sync, and in the end of the movie the end sync, gyro data should reflect that sync to match the video one. Gyroflow should have a sync channel.

see the movie:
[DJI FPV GyroFlow](https://www.youtube.com/watch?v=ErwvWX1IR2E&t=5s)

closed 2022-10-07 by eirasys