VladimirP1/esp-gyrologger #12: Trigger Z-Cam from EspLog

cabinlab, 2022-11-20:
Triggering EspLog from zcam could be possible over wifi.

Check what your browser shows when you navigate to http://\<IP address of z-cam\>/ctrl/mode?action=query

_Originally posted by @VladimirP1 in https://github.com/VladimirP1/esp-gyrologger/issues/2#issuecomment-1321236499_

The Z-Cam can be controlled multiple ways. The DJI RS2 controls it through the COM port. Pretty sure the Zhiyun Weebill S I used to have used the USB-C. The [Flowshutter documentation says it uses UART](https://docs.gyroflow.xyz/flowshutter/clist/#zcam-uart-protocol), but dosn't specify a particular port.

Then there's Ethernet and WiFi. It would be nice to be able to choose from more than one option. For example, I currently have an RS2 connected to the COM, so I can't connect anything else there. And I'm currently using WiFi to stream the video feed. I happen to have the WiFi connecting through a router, so I could put the EspLog on the same network, and that should still work. But maybe there are other use cases where WiFi wouldn't be ideal. I'm not sure.

cabinlab, 2022-11-20:
Over WiFi, this is the JSON from /ctrl/mode?action=query:

`{"code":0,"desc":"","msg":"rec"}`

cabinlab, 2022-11-20:
> There's also a mention of some kind of websocket notification interface, but this https://github.com/imaginevision/Z-Camera-Doc/blob/master/E2/protocol/http.md#Asyc-notification . If you can save the web page from the camera they are talking about, this could be useful

```
CONNECTING: WS://10.98.33.1:81/WWW/HTML/CONTROLLER.HTML

CONNECTION READY

	{
   "what": "TempUpdate",
   "value": 26
}	

	{
   "what": "RecUpdateDur",
   "index": 4,
   "value": 19
}	

	{
   "what": "ConfigChanged",
   "key": "battery_voltage",
   "value": "80"
}	

	{
   "what": "RecUpdateDur",
   "index": 4,
   "value": 20
}	

	{
   "what": "RecUpdateDur",
   "index": 4,
   "value": 21
}	

	{
   "what": "TempUpdate",
   "value": 26
}	

	{
   "what": "RecUpdateDur",
   "index": 4,
   "value": 22
}	

	{
   "what": "RecStarted"
}	

	{
   "what": "RecUpdateRemain",
   "value": 223
}	

	{
   "what": "RecordingFile",
   "value": "A001C1809"
}	

	{
   "what": "RecUpdateDur",
   "index": 4,
   "value": 23
}	

	{
   "what": "RecUpdateDur",
   "index": 8,
   "value": 1
}	

	{
   "what": "RecUpdateDur",
   "index": 1,
   "value": 1
}	

	{
   "what": "RecUpdateDur",
   "index": 4,
   "value": 24
}	

	{
   "what": "RecUpdateDur",
   "index": 8,
   "value": 2
}	

	{
   "what": "RecUpdateDur",
   "index": 1,
   "value": 2
}	

	{
   "what": "TempUpdate",
   "value": 26
}	

	{
   "what": "ConfigChanged",
   "key": "battery_voltage",
   "value": "80"
}	

	{
   "what": "RecUpdateDur",
   "index": 4,
   "value": 25
}	

	{
   "what": "RecUpdateDur",
   "index": 8,
   "value": 3
}	

	{
   "what": "RecUpdateDur",
   "index": 1,
   "value": 3
}	

	{
   "what": "RecUpdateDur",
   "index": 4,
   "value": 26
}	

	{
   "what": "RecUpdateDur",
   "index": 8,
   "value": 4
}	

	{
   "what": "RecUpdateDur",
   "index": 1,
   "value": 4
}	

	{
   "what": "RecUpdateDur",
   "index": 4,
   "value": 27
}	

	{
   "what": "RecStoped"
}	

	{
   "what": "RecUpdateRemain",
   "value": 223
}	

	{
   "what": "TempUpdate",
   "value": 26
}	

	{
   "what": "RecUpdateDur",
   "index": 4,
   "value": 28
}	

CONNECTION CLOSED!
```

cabinlab, 2022-11-20:
Combining the two, `http://host/ctrl/rec?action=start` corresponds with: 
```
	{
   "what": "RecStarted"
}	
```

Then stop recording, `http://host/ctrl/rec?action=stop` with:
```
	{
   "what": "RecStoped"
}	
```

VladimirP1, 2022-11-22:
Oh, cool, so it is possible to get the file name and name the log file accordingly.
But I'll have to think how to make it work and still have the functionality to auto-delete oldest logs.
Probably I can just store the file name in a separate file  or even inside the log file