VladimirP1/esp-gyrologger #13: Custom filenames for logged data with continous filenaming option

surami8, 2022-12-18:
It would be nice, if we could give custom filenames for the logged datas. The best would be eg. an automatic continous filenaming, where we can type in our camera's video filename format and where the logger should start the numbering.

For example my next file name on my camera would be A001-434 (DNG sequence folder) or a A001-434.mov file, so the logger should start here and save the log files as A001-434.BIN.

This way wo could match the videos with the datas easier and if we need to delete a video from our camera during a day long shooting on site, we would immediately know which one BIN should be deleted also, but the naming continous nevertheless in spite of the situation.

VladimirP1, 2022-12-19:
Not sure if there's a point in adding it into the firmware. 
You can easily do this with a mass renaming tool like this one https://doublecmd.github.io/doc/en/multirename.html
For your example just enter A-001-[C].BIN in the 'File name' box and set the start number accordingly.

I can probably add this feature on client side (because the logs are converted from binary form to gcsv in browser anyway) if it would be useful.

surami8, 2022-12-19:
A popup window saves as onclick function would be also good enough on the client side.

VladimirP1, 2022-12-19:
A question about your workflow - is ~50 minutes of logging enough between log downloads? (This is roughly the time I can store in esp32-c3 internal flash).

VladimirP1, 2022-12-19:
If it is not, either esp32 with 16mb flash can be used ( 4 times larger flash ) like this https://sl.aliexpress.ru/p?key=IDIJt4p

I do not currently support external memory, but might theoretically implement it at some point.

surami8, 2022-12-20:
If you implement external SD storage, it's not necessary. It was just a thought.