VladimirP1/esp-gyrologger #14: consider supporting ESP32-S3

gretel, 2023-01-11:
nifty new devices coming like https://shop.m5stack.com/products/atoms3-dev-kit-w-0-85-inch-screen - regards

B-nutze-RR, 2023-01-12:
I'm currently talking to them in order to figure out if we get C3, bmi270, SDcard, button and LED into one product 🤞

cedricchangy, 2023-02-22:
> I'm currently talking to them in order to figure out if we get C3, bmi270, SDcard, button and LED into one product 🤞

that could be awsome! Do you have any news ?

VladimirP1, 2023-05-17:
esp32-s3 support added in https://github.com/VladimirP1/esp-gyrologger/commit/f9b571b87c2600c277576ac3dd148fb833da66d2

closed 2023-05-17 by VladimirP1

AdrianEddy, 2023-05-17:
Pin config for this AtomS3 device:
SDA: 38
SCL: 39
Button: 41

pouletmou, 2024-03-20:
Hello,
would that be possible to display the filename in use when recording on this device ? I believe you did it with the esp oled screen and the m5stick c plus. I usually film on my canon R6II, and I'm often confused about what file goes with my video rush. I thought that if I could have the name when recording I could say it loud and it would be on my video file.

VladimirP1, 2024-03-20:
Yes, that is possible to implement, the display is almost the same, but I do not like the way I have done it for m5stickc. I'll think about it

DinhLe1984, 2025-06-07:
@AdrianEddy and @VladimirP1 Could you advise me the led pin in Atoms3 I should set?

VladimirP1, 2025-06-10:
@DinhLe1984 there's no led on atoms3, but there's a special firmware here https://github.com/VladimirP1/esp-gyrologger/releases/download/v1.1/esplog_esp32s3_atoms3_st7789.zip that abuses as a led if you set "display type" to 4 in settings

FrankMan1-A, 2026-09-05:
Will you ever add support for the M5StickS3 (ESP32S3)?