VladimirP1/esp-gyrologger #15: esp32s

sshahriyari, 2023-03-01:
Hi

I could not find esp32-c3 in my area, And I used esp32s with FTDI programmer.

Everything seemed fine until it came to connecting to the wifi !

I could not find "esplog_vigorous_jang" or anything like this in my wifi list.

VladimirP1, 2023-03-01:
Did you flash the correct firmware ( esplog_esp32, not esplog_esp32c3 ) ?
If you did, try opening a serial console on your computer at 115200, reset the esp32 ( connect EN pin to gnd briefly) and see what it prints

sshahriyari, 2023-03-01:
yes, I used esplog_esp32 firmware.

I did according to this tutorial:
https://electronoobs.com/eng_arduino_tut172.php

where is serial console?
Is there an video tutorial in this regard?

sshahriyari, 2023-03-01:
Arduino serial monitor gives this message:

```
Brownout detector was triggered

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6608
load:0x40078000,len:14788
ho 0 tail 12 room 4
load:0x40080400,len:3792
entry 0x40080694
[0;32mI (29) boot: ESP-IDF v4.4.1 2nd stage bootloader[0m
[0;32mI (29) boot: compile time 08:07:56[0m
[0;32mI (29) boot: chip revision: 1[0m
[0;32mI (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0[0m
[0;32mI (39) boot.esp32: SPI Speed      : 40MHz[0m
[0;32mI (43) boot.esp32: SPI Mode       : DIO[0m
[0;32mI (48) boot.esp32: SPI Flash Size : 4MB[0m
[0;32mI (52) boot: Enabling RNG early entropy source...[0m
[0;32mI (58) boot: Partition Table:[0m
[0;32mI (61) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (84) boot:  2 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (91) boot:  3 storage          Unknown data     01 81 00110000 002f0000[0m
[0;32mI (99) boot: End of partition table[0m
[0;32mI (103) boot_comm: chip revision: 1, min. application chip revision: 0[0m
[0;32mI (110) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=32c80h (208000) map[0m
[0;32mI (194) esp_image: segment 1: paddr=00042ca8 vaddr=3ffb0000 size=08c54h ( 35924) load[0m
[0;32mI (208) esp_image: segment 2: paddr=0004b904 vaddr=40080000 size=04714h ( 18196) load[0m
[0;32mI (216) esp_image: segment 3: paddr=00050020 vaddr=400d0020 size=83bf8h (539640) map[0m
[0;32mI (411) esp_image: segment 4: paddr=000d3c20 vaddr=40084714 size=1405ch ( 82012) load[0m
[0;32mI (445) esp_image: segment 5: paddr=000e7c84 vaddr=50000000 size=00010h (    16) load[0m
[0;32mI (459) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (459) boot: Disabling RNG early entropy source...[0m
[0;32mI (470) cpu_start: Pro cpu up.[0m
[0;32mI (471) cpu_start: Starting app cpu, entry point is 0x400816c0[0m
[0;32mI (457) cpu_start: App cpu up.[0m
[0;32mI (487) cpu_start: Pro cpu start user code[0m
[0;32mI (487) cpu_start: cpu freq: 160000000[0m
[0;32mI (487) cpu_start: Application information:[0m
[0;32mI (492) cpu_start: Project name:     esp-gyrologger[0m
[0;32mI (497) cpu_start: App version:      1[0m
[0;32mI (502) cpu_start: Compile time:     Feb  5 2023 08:07:49[0m
[0;32mI (508) cpu_start: ELF file SHA256:  4d39e52868d70784...[0m
[0;32mI (514) cpu_start: ESP-IDF:          v4.4.1[0m
[0;32mI (519) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (526) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (532) heap_init: At 3FFBE788 len 00021878 (134 KiB): DRAM[0m
[0;32mI (538) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (544) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (551) heap_init: At 40098770 len 00007890 (30 KiB): IRAM[0m
[0;32mI (558) spi_flash: detected chip: generic[0m
[0;32mI (562) spi_flash: flash io: dio[0m
[0;32mI (567) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
I (787) wifi:wifi driver task: 3ffc8a10, prio:23, stack:6656, core=1
[0;32mI (787) system_api: Base MAC address is not set[0m
[0;32mI (787) system_api: read default base MAC address from EFUSE[0m
I (787) wifi:wifi firmware version: 63017e0
I (797) wifi:wifi certification version: v7.0
I (797) wifi:config NVS flash: disabled
I (797) wifi:config nano formating: disabled
I (807) wifi:Init data frame dynamic rx buffer num: 32
I (807) wifi:Init management frame dynamic rx buffer num: 32
I (817) wifi:Init management short buffer num: 32
I (817) wifi:Init dynamic tx buffer num: 32
I (827) wifi:Init static rx buffer size: 1600
I (827) wifi:Init static rx buffer num: 10
I (827) wifi:Init dynamic rx buffer num: 32
[0;32mI (837) wifi_init: rx ba win: 6[0m
[0;32mI (837) wifi_init: tcpip mbox: 32[0m
[0;32mI (847) wifi_init: udp mbox: 6[0m
[0;32mI (847) wifi_init: tcp mbox: 6[0m
[0;32mI (847) wifi_init: tcp tx win: 5744[0m
[0;32mI (857) wifi_init: tcp rx win: 5744[0m
[0;32mI (857) wifi_init: tcp mss: 1440[0m
[0;32mI (867) wifi_init: WiFi IRAM OP enabled[0m
[0;32mI (867) wifi_init: WiFi RX IRAM OP enabled[0m
[0;32mI (877) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07[0m
[0;33mW (877) phy_init: failed to load RF calibration data (0xffffffff), falling back to full calibration[0m
```

VladimirP1, 2023-03-01:
try checking your power supply to esp32 --- brownout usually means power supply problems
I am very busy atm will check again later if the brownout threshold is set to the lowest possible level for esp32

sshahriyari, 2023-03-01:
I added 10k pullup resistors between TX and 3.3V (esp32), also between GPIO0 and GND.
Could it be because of this?

Thank you for your follow up.

sshahriyari, 2023-03-03:
> try checking your power supply to esp32 --- brownout usually means power supply problems I am very busy atm will check again later if the brownout threshold is set to the lowest possible level for esp32

I'm wating for your response :)

VladimirP1, 2023-03-03:
Yeah, the pullup resistors might be the reason of the message, this explains why it is in the beginning.
I have checked - the brownout detector is configured to the lowest possible voltage - 2.43 volts. 

Can you try full erase `esptool.py erase_flash` and reflash to be sure the NVS region is erased?

VladimirP1, 2023-03-03:
How can you do what? 
Erase flash?
Run the command I posted in the directory with esptool, then run the flashing command from the readme in the artifact zip file.

VladimirP1, 2023-03-03:
And maybe it is a good idea to try a wifi hotspot example before that. Just to make sure there is enough power for wifi to calibrate

sshahriyari, 2023-03-03:
I checked the voltage with multimeter, it's around 2.84 volts (esp32 side).

VladimirP1, 2023-03-03:
That's too low - because esp32 consumes high current during wifi initialization (hundreds of milliamps for about 1 millisecond). This causes an very short voltage drop if your voltage source is not powerful ebough which you won't notice without an oscilloscope but it will trigger brownout.

sshahriyari, 2023-03-03:
Currently the ESP32 is powered via USB. How can I increase the voltage?
I can set the FTDI to 5V but I'm afraid the ESP32 will burn!

VladimirP1, 2023-03-03:
Can you post a picture of your setup? It seems that you are trying to power the esp32 from FTDI's internal 50ma voltage regulator which is too weak for that.

sshahriyari, 2023-03-03:
where I can post picture?
do you have telegram?

VladimirP1, 2023-03-03:
Yes: `@vpinch`

sshahriyari, 2023-03-03:
I fixed voltage (3.2v)

now I got these messages:

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57
```

sshahriyari, 2023-03-04:
I uploaded "AsyncUDPServer.ino" to esp32 and serial monitor gives me this message :

```
Brownout detector was triggered

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13924
ho 0 tail 12 room 4
load:0x40080400,len:3600
entry 0x400805f0
```

sshahriyari, 2023-03-04:
Is it possible that esp32 is burnt?

sshahriyari, 2023-03-05:
I fixed the "Brownout" problem with using external power source.

now uploaded "AsyncUDPServer" arduino esp32 example, and it gives me this message:

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13924
ho 0 tail 12 room 4
load:0x40080400,len:3600
entry 0x400805f0
WiFi Failed
```

VladimirP1, 2023-03-05:
Try "WiFiAccessPoint". Because WiFi failed means connection failed, this can be due to many reasons, for example wrong frequency band (if you have 5ghz AP) or wrong password

closed 2023-03-06 by VladimirP1