VladimirP1/esp-gyrologger #20: Experience and Feedback Using esp-gyrologger on M5StickC Plus

gmag11, 2023-11-24:
Hi. I've recently come across both Gyroflow and the esp-gyrologger project and I must say, I am thoroughly impressed by these extraordinary works. Kudos to everyone involved!

I happen to own an M5StickC Plus module and thought it would meet the requirements for using with your firmware. Here are some specifics about the M5StickC Plus that I found beneficial:

- **ESP32** Processor with **4MB** flash.
- **MPU6886 Accelerometer**: Thanks for including support for this!
- **LED** and **2 Buttons**: Handy for interactions.
- **Battery**: Makes it conveniently portable.
- **ST7789v2 Display** (135x240 resolution): Great for visual feedback, although not yet supported.
- Robust **Plastic Casing**: Durable and reliable.
- **Affordable Price Tag** (~$20): Makes it accessible for many.

[Detailed specs](https://docs.m5stack.com/en/core/m5stickc_plus)

My initial tests have been quite positive. The accelerometer, LED, and WiFi communication are all functioning correctly, and the calibration process went smoothly. I haven't tested it with Gyroflow yet, but I'm looking forward to it.

One thing I noticed is that the ST7789 display isn't currently supported. It would be fantastic to have this feature in the future, as the display can provide valuable real-time feedback.

I wanted to share my experience because I believe using this module could be a more straightforward option for many, compared to building a custom one. It offers a great balance of features, convenience, and cost-effectiveness.

Looking forward to any future updates and will be more than happy to provide further feedback if needed!

VladimirP1, 2023-11-24:
I wrote some hacky (it is a wrapper around esp_lcd which adds support for it into u8g2 in monochrome mode) code, which I do not really like for st7789 about a month ago. I just pushed it on `lcd_st7789` branch. Probably will provide binaries later.

VladimirP1, 2023-11-24:
This https://www.youtube.com/shorts/K1KbeRfWatg

VladimirP1, 2023-11-24:
Added binaries here https://github.com/VladimirP1/esp-gyrologger/releases/tag/v1.1

Use  `esplog_m5stickcplus_st7789.zip` and set "display type" to 3.

closed 2023-11-25 by VladimirP1