VladimirP1/esp-gyrologger #25: Hello

igorbmn, 2024-10-30:
When i try to flash the firmware, i get this error: 

A fatal error occurred: Failed to connect to Espressif device: No serial data received.

igorbmn, 2024-10-30:
I was able to flash the firmware and connect to the wifi signal but i don't know how to do this step: 

Go to http://192.168.4.1/settings, enter the SDA and SCL pin numbers you have connected the IMU to, click "Apply".

I'm using a simple usb cable connected to my computer.

VladimirP1, 2024-10-31:
You do not know the pin numbers? Or what is your problem with this exactly?

igorbmn, 2024-10-31:
Thank you for the support. I just used a regular usb c cable, didn't mod it. I don't know the pin numbers. I'm using a regular cable via the USB input o my atom s3.

VladimirP1, 2024-10-31:
https://github.com/VladimirP1/esp-gyrologger/issues/14#issuecomment-1552180153

igorbmn, 2024-10-31:
It worked, thank you very much, my friend!

closed 2024-10-31 by igorbmn