VladimirP1/esp-gyrologger #26: Working

igorbmn, 2024-10-31:
Hello, I was able to make it work but I saw a video of a guy using it and his atom s3 blinks green when turned on, and blinks red when recording. How can I do that with mine?

VladimirP1, 2024-10-31:
Use firmware from `esplog_esp32s3_atoms3_st7789.zip` (if not already) and set in settings "Display type" to 4.

closed 2024-10-31 by VladimirP1

igorbmn, 2024-10-31:
Do I need to uninstall the the other firmware before flashing this one?

VladimirP1, 2024-10-31:
No, but if you have problems you can do `esptool erase_flash`

igorbmn, 2024-10-31:
It works, thank you again. 

One last thing: how do I calibrate it? 

Sorry to ask so many questions, I newby to this world.

VladimirP1, 2024-10-31:
There is gyro calibration (you just press the green round button on the webpage while the device is stationary). It is saved to flash.
Also there is accel calibration. This is how to do it https://youtu.be/X76fzDprrrg?si=z-T7vP07ozyeVk4C