VladimirP1/esp-gyrologger #30: Option to download &/or delete all files at once

handcraftedcc, 2025-03-05:
Would be nice if the web interface had the option to download and then remove all stored files at once, instead having to click download and then delete for each one...

VladimirP1, 2025-03-05:
Yeah, but I am not planning to implement this
I am ready accept PRs and maybe fix something critical, not more

With current design of esp-gyrologger a quick and dirty way of implementing download could be on client side (in js code which downloads the compressed file and runs the WASM-compuled decompressor). SD card formatting functionality is available but it is hidden from user (can be triggered by doing a GET request on `/format`)

The thing is that I do not really like how I designed the firmware in many ways (the choice of esp http server which can only process one http request at a time, the choice of the fat32 filesystem which causes issues when there are like >200 files when logging to external SD card, too complex gyro data interpolation code, lack of any header on the compressed files so that they cannot be identified and loaded directly into gyroflow without conversion to gcsv). So in reality it would be nice to rewrite the firmware again from scratch considering the things I learnt, but I am not motivated enough for that. On the other hand, I do not really want to add new features to the current version because I do not like many things about it.

handcraftedcc, 2025-03-05:
Oh gotcha. That's a bummer.
Been working on a custom hardware build and your toolset has so much
potential. Would be amazing to have an update like you described...