VladimirP1/esp-gyrologger #31: AtomS3 SDA/SCL

belamauro, 2025-03-10:
Hi :)

 First of all and just to say it in advance, I am new to all this and I'm sorry if I ask stupid questions.

I have managed to flash my AtomS3 and AtomS3-Lite with the firmware. 
I recieved the wifi signal and wanted to setup the SDA and SCL Pins within the settings.

The problem is that I don't know which SDA or SCL Pins either of these controller uses.
I tried to google it and tried a lot of combinations, but none of them seem to work.
Does someone know what values I have to enter for the SCL and SDA Pins?

Also I think that I have killed my AtomS3, because it suddenly stopped sending out the WIFI signal and is also not showing anything on the display. 
It tuns on but thats about it. Is there a way to solve this?

Thank you for your help, I really don't know what I'm doing. haha

VladimirP1, 2025-03-10:
https://github.com/VladimirP1/esp-gyrologger/issues/14#issuecomment-1552180153

VladimirP1, 2025-03-10:
The reason it has stopped working is that you probably reconfigured one of the internal spi flash pins as i2c so that it cannot execute the firmware from that flash. To fix your atoms3 use `esptool erase_flash` and then reflash the firmware.

closed 2025-03-10 by VladimirP1

belamauro, 2025-03-10:
I will do that, thank you very much!
Do you know to which pins I should set the SDA and SCL? 
It wasn't 6&7, also not 16&17 or 8&9.

VladimirP1, 2025-03-10:
See here https://github.com/VladimirP1/esp-gyrologger/issues/14#issuecomment-1552180153

belamauro, 2025-03-10:
Ups, didn't see it earlier. Thank you so much! I hope I can get it to work now.