VladimirP1/esp-gyrologger #32: Logger is logging strange numbers

dnvn3000, 2025-04-24:
Here is a sample of data that I'm getting from esp-logger using the esplog_esp32s3_atoms3_st7789.zip release. Everything else seems to be working fine. This strange number "32438" keeps popping up and my data is not correct when I bring it into gyroflow. 

`GYROFLOW IMU LOG
version,1.1
id,esplog
orientation,xyz
tscale,0.00180
gscale,0.00053263221
ascale,0.00048828125
t,gx,gy,gz,ax,ay,az
1,-32438,32438,32438,-142,-411,-2160
2,32438,0,32438,-142,-411,-2160
3,0,0,32438,-142,-411,-2160
4,0,0,32438,-142,-411,-2160
5,0,32438,32438,-142,-411,-2160
6,0,0,32438,-142,-411,-2160
7,0,0,32438,-142,-411,-2160
8,0,32438,32438,-142,-411,-2160
9,0,0,32438,-142,-411,-2160
10,0,0,32438,-142,-411,-2160
11,0,32438,32438,-142,-411,-2160
12,0,0,32438,-142,-411,-2160
13,0,0,32438,-142,-411,-2160
14,0,32438,32438,-142,-411,-2160
15,0,0,32438,-142,-411,-2160
16,32438,0,32438,-142,-411,-2160
17,0,32438,32438,-142,-411,-2160
18,0,0,32438,-142,-411,-2160
19,0,0,32438,-142,-411,-2160
20,0,0,32438,-142,-411,-2160
21,0,32438,32438,-142,-411,-2160
22,0,0,32438,-142,-411,-2160
23,0,0,32438,-142,-411,-2160
24,0,32438,32438,-142,-411,-2160
25,0,0,32438,-142,-411,-2160
26,32438,0,32438,-142,-411,-2160
27,0,32438,32438,-142,-411,-2160
28,0,0,32438,-142,-411,-2160
29,0,0,32438,-142,-411,-2160
30,0,32438,32438,-142,-411,-2160
31,0,0,32438,-142,-411,-2160
32,0,0,32438,-142,-411,-2160
33,0,32438,32438,-142,-411,-2160
34,0,0,32438,-142,-411,-2160
35,32438,0,32438,-142,-411,-2160
36,0,32438,32438,-142,-411,-2160
37,0,0,32438,-142,-411,-2160
38,0,0,32438,-142,-411,-2160
39,0,32438,32438,-142,-411,-2160
40,0,0,32438,-142,-411,-2160
41,0,0,32438,-142,-411,-2160
42,0,0,32438,-142,-411,-2160
43,0,32438,32438,-142,-411,-2160
44,0,0,32438,-142,-411,-2160
45,32438,0,32438,-142,-411,-2160
46,0,32438,32438,-142,-411,-2160
47,0,0,32438,-142,-411,-2160
48,0,0,32438,-142,-411,-2160
49,0,32438,32438,-142,-411,-2160
50,0,0,32438,-142,-411,-2160
51,0,0,32438,-142,-411,-2160
52,0,0,32438,-142,-411,-2160
53,0,32438,32438,-142,-411,-2160
54,0,0,32438,-142,-411,-2160
55,0,0,32438,-207,-259,-1764
56,32438,32438,32438,-207,-259,-1764
`

VladimirP1, 2025-04-24:
Is it the whole log like this or only the beginning?
if it is the whole log, can you check if the gyro chip works properly in arduino by flashing this example https://github.com/m5stack/M5AtomS3/blob/main/examples/Basics/imu/imu.ino ?

dnvn3000, 2025-04-24:
It's the first 255 entries in my log.

D

VladimirP1, 2025-04-24:
Send the full log

dnvn3000, 2025-04-24:
Here is a screen shot of what the graph looks like in gyroflow from a
previous test. I would expect to see all the lines more or less around the
same level. I've also attached the entire log.

[image: image.png]

Donovan

dnvn3000, 2025-04-24:
[LAA00001.BIN.gcsv.txt](https://github.com/user-attachments/files/19898469/LAA00001.BIN.gcsv.txt)

[image]

attached log and image from previous post

Donovan

VladimirP1, 2025-04-24:
Try gyro calibration (round green button in web ui) when the device is not moving. If that does not help, that's probably a faulty gyro, but to know for sure, you can oad that arduino sketch I mentioned above and see if it behaves the same there.

dnvn3000, 2025-04-24:
[LAA00003.BIN.gcsv.txt](https://github.com/user-attachments/files/19898914/LAA00003.BIN.gcsv.txt)

Things are looking better. I plotted with chatGPT and its looking good. Will test tonight with gyroflow. 

[image]

Thank you!

D